balena-io-experimental/balena-electronjs #1: Error: Running without the SUID sandbox!

adamj88, 2016-04-13:
I'm experiencing an error whilst trying to run this on a Raspberry Pi 3, not sure if this is the same for the 2. The error I'm seeing is as follows:

```
13.04.16 21:51:10 [+0100] [75:0413/205110:ERROR:browser_main_loop.cc(217)] Running without the SUID sandbox! See https://chromium.googlesource.com/chromium/src/+/master/docs/linux_suid_sandbox_development.md for more information on developing with the sandbox on.
13.04.16 21:51:23 [+0100] [75:0413/205123:ERROR:browser_gpu_channel_host_factory.cc(145)] Failed to create channel.
```

Please let me know if there is any more information you need to debug this.

curcuz, 2016-04-15:
hi, Those errors shouldn't prevent the execution of your app (i'm investigating them anyways). 

How did you set the env vars?

dovy, 2016-05-25:
Similar:
[77:0525/040715:ERROR:browser_main_loop.cc(217)] Running without the SUID sandbox! See https://chromium.googlesource.com/chromium/src/+/master/docs/linux_suid_sandbox_development.md for more information on developing with the sandbox on.

Can't get the gui to launch and running the direct IP address yields that nothing is running on 80.

Any ideas? Raspberry Pi.

curcuz, 2016-05-25:
this is because i made the root user start electron. As a template purpose, this is good since people tinkering/learning with GPIO or other interfaces won't get stuck on permission issues

dovy, 2016-05-26:
@curcuz what's the suggested method to do this properly?  ;)

curcuz, 2016-05-26:
@dovy apparently, just the way it is already 😄  https://github.com/electron/electron/issues/1482

closed 2016-06-09 by curcuz