balena-io-experimental/balena-electronjs #2: Enable mouse

dovy, 2016-05-25:
Got it to boot after changing some things around. Great work. For development, how would one enable the mouse in this interface? I'd like to use it until I hook up a touch-screen.  ;)

curcuz, 2016-05-25:
you'd want to remove the `-nocursor` parameter from [here](https://github.com/fwrgit/resin-electronjs/blob/d968e602967f25e0009bf37dbe9846e1dfb76fd2/Dockerfile.template#L48)

dovy, 2016-05-25:
Are the environment variables from resin accessible in the dockerfile? Could be another option to "ENABLE_MOUSE".

;)

closed 2016-05-25 by dovy

curcuz, 2016-05-25:
There is indeed a way to access env vars from dockerfile, but that would be effective only once during the push, then until the next push changes to variables won't be reflected to the application

dovy, 2016-05-25:
Hmm, true. Just wanted to be able to modify easily.

dovy, 2016-05-25:
Completely off this topic, but how did you alter the boot to not include the Resin boot screen while X is launching?

dovy, 2016-05-25:
meaning the black screen with the Resin in ASCII, not the boot loader graphic.  ;)

curcuz, 2016-05-26:
@dovy definitely not related to this app :) to achieve that atm you have to tamper the hostOS by deleting `etc/systemd/system/multi-user.target.wants/tty-replacement.service` but hopefully in the future there might be some kind of switch to do that more easily

mateodelnorte, 2018-09-26:
@curcuz removing `-nocursor` allows us to see a cursor, but the cursor doesn't move when we move the mouse. We can see mouse events using `uvtest`, within the container. They're just not making it into X11 or the electron app. 

Any ideas?

g3rrydanc3, 2019-05-04:
For future people who have same problem like @mateodelnorte 

1. Remove the -nocursor parameter from [here](https://github.com/balena-io/resin-electronjs/blob/d968e602967f25e0009bf37dbe9846e1dfb76fd2/Dockerfile.template#L48)
2. Add `ENV UDEV=1` to [here](https://github.com/balena-io/resin-electronjs/blob/d968e602967f25e0009bf37dbe9846e1dfb76fd2/Dockerfile.template#L2)
3. Add to `systemd \` [here](https://github.com/balena-io/resin-electronjs/blob/d968e602967f25e0009bf37dbe9846e1dfb76fd2/Dockerfile.template#L42)

4. (optional) 
> Generally for systemd, it just requires installing the systemd package, masking a number of services and defining a new entry.sh and a resin.service.
> 
> From [Docs ](https://www.balena.io/docs/reference/base-images/base-images/#working-with-dynamically-plugged-devices)

Suggest add this code to [here](https://github.com/balena-io/resin-electronjs/blob/d968e602967f25e0009bf37dbe9846e1dfb76fd2/Dockerfile.template#L44)
```
RUN systemctl mask \
    dev-hugepages.mount \
    sys-fs-fuse-connections.mount \
    sys-kernel-config.mount \

    display-manager.service \
    getty@.service \
    systemd-logind.service \
    systemd-remount-fs.service \

    getty.target \
    graphical.target
```