balena-io-experimental/balena-electronjs #5: Enable audio input and output

edoardo849, 2016-09-04:
On my device(s) I need to access the microphone and output audio in kiosk mode. However, when I boot up the app it appears to be stucked and in the logs I see 2 messages:

04.09.16 12:18:31 [+0100] [104:0904/111831:INFO:CONSOLE(170)] "> ON EVENT", source: file:///app/app/app.js (170)
04.09.16 12:18:31 [+0100] [104:0904/111831:INFO:CONSOLE(171)] "1", source: file:///app/app/app.js (171)
04.09.16 12:18:31 [+0100] [104:0904/111831:INFO:CONSOLE(172)] "Waiting for approval to access your microphone ...", source: file:///app/app/app.js (172)
04.09.16 12:18:31 [+0100] [104:0904/111831:INFO:CONSOLE(163)] "> ON ERROR", source: file:///app/app/app.js (163)
04.09.16 12:18:31 [+0100] [104:0904/111831:INFO:CONSOLE(164)] "5", source: file:///app/app/app.js (164)
04.09.16 12:18:31 [+0100] [104:0904/111831:INFO:CONSOLE(165)] "No live audio input in this browser: {}", source: file:///app/app/app.js (165)

so basically it seems to be waiting for the user's permissions (that I cannot have in kiosk mode) and cannot find an audio input device. Any help?

curcuz, 2016-09-05:
Hi @edoardo849 can you please paste a snippet of code to reproduce the issue? I have some suspects and want to be sure about it

edoardo849, 2016-09-05:
Hey @curcuz ! Thanks for your quick reply, the mic that I was using wasn't working. Looks like the error was thrown when no mic was detected. I haven't tested input and audio output yet but the aforementioned error is solved.

curcuz, 2016-09-05:
@edoardo849 I'll close this one then, feel free to open new ones if you encounter any other issues!

closed 2016-09-05 by curcuz

edoardo849, 2016-09-17:
Hello @curcuz ! So after several trials I'm still having issues with audio. Input audio works with your default Dockerfile but audio does not. Now if I install `omxplayer` I can play audio from the terminal. However, still no luck in Electron. If I install `alsa-utils` and set up `/etc/asound.conf` like this:

```txt
pcm.!default {
	type hw
	card 1
}

ctl.!default {
	type hw           
	card 0
}
```

the audio works fine on Electron, but the input (microphone) does not. I guess my short question is: how do I get audio input and output to work on Electron?