balena-io-experimental/balena-electronjs #7: gutsubmodules

klowdo, 2016-09-06:
I want to do this myself but i don't know how. i need it do download gitsubmodules

curcuz, 2016-09-06:
@klowdo - do you mean on build process via Dockerfile or just in your fork?

klowdo, 2016-09-06:
in the docker file. i'm using atlasboard that use submodules as plugins to the dashboard. i tried adding git-core to docker file. but the "git repository" don't exsist in src

curcuz, 2016-09-07:
@klowdo I don't understand this statement:

>  but the "git repository" don't exsist in src

can you please clarify?

klowdo, 2016-09-09:
@curcuz i want to download the git submodules in the docker build. Because i need it in my output directory
I'm using atlasboard and submodule https://bitbucket.org/atlassian/atlasboard-atlassian-package

curcuz, 2016-09-09:
@klowdo 

A git submodule, as per [git official documentation](https://git-scm.com/docs/git-submodule) is a way "to keep another Git repository in a subdirectory of your repository." - so you should either add the desired repos as submodules to your own repo or clone them on the Dockerfile.

So, again, I don't understand this statement:
> but the "git repository" don't exsist in src

what src?

Also, 

>  i want to download the git submodules in the docker build

you can't *download* submodules since they are just links to repositories. You can *clone* them in the docker build 

Maybe you meant that the resin builder ignores submodules you have set in your repository? in that case, can you please provide a description of how you added the submodules before pushing to resin builder, what is expected and what is the given output from the builder?

klowdo, 2016-09-10:
Okey i understand! i wll clone in advance an push them in as my own code! thanks!

closed 2016-09-10 by klowdo