balena-io-experimental/balena-electronjs #9: Video performance

manast, 2016-09-12:
Hello,

I have been checking resin-electronjs today and I am not able to get any decent performance on a raspberry pi 3. I have tried with the master branch as well as with the framebuffer one. Also, I noticed that the image quality is not so good, it seems as if the graphics are quantized, i.e. running in maybe 16bits/pixel instead of 24. Do you have any advices? is it even possible to get an acceptable performance to play videos using electron on a rpi?

mrjonny2, 2016-09-30:
I have attempted to get this working, unfortunately chromium (on which electron is based) does not support the GPU used in the RPi, this means that it is unable to use hardware acceleration and the full load is dumped onto the CPU, which does not handle video decoding very well.

buildog, 2016-10-25:
maybe by activating WebGL but don't know how to do that within a resin image ?

TomAshley303, 2017-01-12:
This is actually my main use case for wanting to buy a raspberry pi: An electron app meant for video playback.

danielmahon, 2017-01-12:
@TomAshley303 In my experience, until they update to the 4.4 kernel which has the (experimental) OpenGL support you will pretty much HAVE to rely on omxplayer for video playback. I know the resin team is working on the update as we speak. For now, I am running an electron app based on this template and then triggering videos via `omx-manager`. I just send button presses/states through the electron IPC from the renderer to the main. Also, since you can set a window size for the omxplayer instance, I just overlay the video on PART of the ui and then you can still use button presses on the page to control the video.

mexpolk, 2017-08-24:
Hi, any updates on OpenGL support?

mrjonny2, 2017-08-30:
It seems to be on chromium's side with a lack of support for mali GPUs on linux

kristjanmik, 2017-10-02:
>I know the resin team is working on the update as we speak
@danielmahon Do you know anything about the progress on the Resin team side?

curcuz, 2017-10-02:
Hello, it turns out the issue is with the mesa libraries that are installed by `debian:jessie`. The way to fix this is to update these packages.
To do that add the jessie-backports mirror to your source.list file, by adding the following line to your Dockerfile:
`RUN echo "deb http://ftp.debian.org/debian jessie-backports main" >> /etc/apt/sources.list`
Lastly use apt-get with `-t jessie-backports` flag to install from that source:
`apt-get -t jessie-backports install {packages}`

more info here https://github.com/resin-os/resin-up-board/issues/26

curcuz, 2017-10-02:
current plan on the resin-electronjs side is to include this workaround on the Dockerfile and release a new version asap

kristjanmik, 2017-10-03:
@curcuz Thanks for the quick reply. This only works on Intel based boards, not the Raspberry Pi 3(which is ARM based). Is there any solution that you know of for ARM based devices?

MiluchOK, 2017-10-26:
I need this working too... Is there anything?

SteffenDE, 2018-01-03:
Maybe someone with enough experience in building electron/chromium could apply one of these patches and build a custom electron binary for the rpi: https://github.com/raspberrypi-ui/chromium_patches
The Chromium version shipped in the latest raspbian releases has one of these patches applied and plays 1080p H264 just fine. Sadly it crashes when using the OpenGL driver, so either you have accelerated video OR accelerated animations and webgl.

apgiorgi, 2018-01-09:
Adding `dtoverlay=vc4-fkms-v3d` into config.txt helped with WebGL/CSS HW performance on Raspberry Pi 3.

Addflow, 2018-08-15:
i'm experiencing the same problems with the rpi3 using the video tag and a 1080p mp4 video.

omxplayer does a decent job and the playback is flawless. but since i want an overlay (text, images, ..) on top of the video, omxplayer won't work for me - omxplayer OR html content :/

does someone know if another board is capable of smooth 1080p video playback using the html5 video tag? what about the intel nuc? thanks!

SteffenDE, 2018-08-15:
@Addflow I‘m using a Celeron NUC with Ubuntu 16.04 and Chromium and it plays 1080p H264 fine.

cyraid, 2018-12-05:
Has chromium fixed this yet? Is it a problem with chromium?

nicshackle, 2019-03-25:
I'm trying to solve this, too. Adding `dtoverlay=vc4-fkms-v3d` did help considerably, but not enough for 1080p. 

I briefly tried to use omxplayer, but battled to install it. @danielmahon can you share how you went about using omxplayer?

g3rrydanc3, 2019-10-15:
I follow tutorial on this site https://thecodersblog.com/hardware-accleration-in-linux-with-chrome/
The package `i965-va-driver` not found, so I ignore it.

This can increase performance, not much but still.
Tested on latest electron@3, raspberry pi 3b+