balena-io-experimental/balena-electronjs #10: Audio not working on Electron

edoardo849, 2016-09-17:
Hello @curcuz  ! So after several trials I'm still having issues with audio. Input audio works with your default Dockerfile but audio does not. Now if I install omxplayer I can play audio from the terminal. However, still no luck in Electron. If I install alsa-utils and set up /etc/asound.conf like this:

pcm.!default {
    type hw
    card 1
}

ctl.!default {
    type hw           
    card 0
}
the audio works fine on Electron, but the input (microphone) does not. I guess my short question is: how do I get audio input and output to work on Electron?

curcuz, 2016-09-23:
Hi @edoardo849 - I think whatever is the issue, it's not related with Electron itself, but with X11-based applications - can you give me details about what device are you using, what's the USB mic model?

curcuz, 2016-10-30:
@edoardo849 since I am unable to reproduce without the info requested, I will close this issue soon if no details will be shared

curcuz, 2016-12-01:
no answer since more than 2 months ago, closing

closed 2016-12-01 by curcuz