balena-io-experimental/balena-electronjs #12: xinit: connection to X server lost

joeyvandijk, 2016-09-30:
When I deploy the code to a Raspberry Pi v1 or a Zero (with 64GB) I see in the Resin logs window X.org being launched and again and again and again at both devices with the same code. I did not alter the code of this repo, just pushed it to the Resin Git repo to deploy and see how it worked.

I see this over and over again:

```txt
30.09.16 14:19:05 [+0200] X.Org X Server 1.17.2
30.09.16 14:19:05 [+0200] Release Date: 2015-06-16
30.09.16 14:19:05 [+0200] X Protocol Version 11, Revision 0
30.09.16 14:19:05 [+0200] Build Operating System: Linux 3.8.11-2-ARCH armv7l Raspbian
30.09.16 14:19:05 [+0200] Current Operating System: Linux raspberrypi-f2686cf 4.1.21 #1 Fri Jul 29 13:32:46 UTC 2016 armv6l
30.09.16 14:19:05 [+0200] Kernel command line: dma.dmachans=0x7f35 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708.boardrev=0x900093 bcm2708.serial=0xad78d864 smsc95xx.macaddr=B8:27:EB:78:D8:64 bcm2708_fb.fbswap=1 bcm2708.uart_clock=3000000 bcm2708.disk_led_gpio=47 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000 dwc_otg.lpm_enable=0 console=null root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
30.09.16 14:19:05 [+0200] Build Date: 11 January 2016 01:32:44PM
30.09.16 14:19:05 [+0200] xorg-server 2:1.17.2-1+rpi1 (http://www.debian.org/support)
30.09.16 14:19:05 [+0200] Current version of pixman: 0.33.3
30.09.16 14:19:05 [+0200] Before reporting problems, check http://wiki.x.org
30.09.16 14:19:05 [+0200] to make sure that you have the latest version.
30.09.16 14:19:05 [+0200] Markers: (--) probed, (**) from config file, (==) default setting,
30.09.16 14:19:05 [+0200] (++) from command line, (!!) notice, (II) informational,
30.09.16 14:19:05 [+0200] (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
30.09.16 14:19:05 [+0200] (==) Log file: "/var/log/Xorg.0.log", Time: Fri Sep 30 12:19:05 2016
30.09.16 14:19:05 [+0200] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
30.09.16 14:19:07 [+0200] xinit: connection to X server lost
30.09.16 14:19:08 [+0200]
30.09.16 14:19:08 [+0200] waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
```

Updating docker-prebuild did not matter anything, but is it Fluxbox, X.org? 
The device is recognised (so it has Ethernet), shows on my TV ("Booted - check your dashboard") but does not show my application. 

What can I try to fix this or is it a bug?

curcuz, 2016-09-30:
Have you set `gpu_mem` to a value > 64 as per the README?

alexandrosm, 2016-09-30:
carlo, may be worth detecting the error and warning the user?

curcuz, 2016-09-30:
@alexandrosm I can for the rpi family, this will inyroduce device-specific code that will be executed on some sort od hw detection clause #3

alexandrosm, 2016-09-30:
hmm, interesting. Is there some sort of cross-device way to find out the
GPU ram? I guess not, right?

joeyvandijk, 2016-10-01:
@curcuz yes I have, but not in the ``ENVIRONMENT VARIABLES`` but in the device configuration because ``RESIN_`` variables cannot be set at ``https://dashboard.resin.io/apps/126733/devices/227742/envvars``. But it does not matter for this issue. Same issue still exists, so keeps repeating the error ``xinit: connection to X server lost``

thrandre, 2016-11-11:
I'm experiencing the exact same issue. RESIN_HOST_CONFIG_gpu_mem is set to `70`

curcuz, 2016-12-15:
#11 should avoid this kind of confusion

joeyvandijk, 2017-01-29:
Trying again after some changes I noticed, but still no luck. I already changed the settings as you all mentioned above, and it did not work.

With the new changes from last weeks I still see no change. The startup process still hangs and after a while it concludes with ` xinit: connection to X server lost`. But this time I see before X.org launches `xauth: file /root/.Xauthority does not exist` in the resin.io startup logs.

But when I use the terminal and log into the device I see /root/.Xauthority there but it is empty. So maybe a race condition? I know that sounds strange, but the rpi-Zero is not as powerful as others.

I tried to display all *DISPLAYS* but none were found, which could be the issue.

So to repeat my issue: I can run the code from this repository, but after booting up the application I still do **not** see any screen/window with Electron showing something. It hangs at the startx/xorg process while it cannot detect a display perhaps, but I cannot find more information what is wrong.

Any ideas with this additional information, because #11 did not help on the rpi-zero.

curcuz, 2017-01-29:
@joeyvandijk are yu able to see the splash logo and the 'resin.io - check your dashboard' message on the screen when the device boots? in general, are you able to get any kind of output on the screen?

joeyvandijk, 2017-01-29:
Yes I can see that, but from that moment nothing happens on the screen and
in the log I see the above error happening after the splash logo and "
resin.io - check your dashboard" message.

curcuz, 2017-01-30:
@joeyvandijk I will try to reproduce - can you confirm the following:

* you are using the latest from `master`
* you are targeting a Raspberry Pi zero

joeyvandijk, 2017-01-30:
Both are `yes`. I started the project yesterday with git clone so yes I
have the latest from master. And this is about the Raspberry Pi Zero.

curcuz, 2017-01-30:
I managed to reproduce. The issue is with Electron instantly exiting, making X end the session as a consequence

bjarneheden, 2017-03-14:
Hi!
Any update/progress on this issue?
We are having what looks like the same issue, but with a Raspberry Pi 3

Aubron, 2017-05-18:
Any thoughts on how to circumvent this? I get it pretty regularly on my board:

http://i.imgur.com/L6E2KCd.png

curcuz, 2017-06-06:
@bjarneheden @Aubron can you share the config variables you set, if different from what documented? (note: **config** vars, not env vars)

Aubron, 2017-06-06:
@curcuz - My apologies, but this was long enough ago that I moved to a different project, and no longer have a test setup.

As far as I am aware, however, I interacted with the project only through environment variables, continuing to try different RESIN_HOST_CONFIG_gpu_mem values to run a datadog dashboard page.

curcuz, 2017-06-06:
@Aubron `RESIN_HOST_CONFIG_gpu_mem` shouldn't be set on environment variables pane, but instead on the configuration variables one. You also get a feedback in the logs, since the supervisor tells you it's applying the new config - if you have any chance to confirm this, it will help me a lot debugging an issue that so far I was only able to reproduce on the Pi Zero.

Thanks!

Aubron, 2017-06-06:
I'm ordering a new pi next week, when it comes in I'll see if I can't set this back up and repro, just don't want to tear apart my finished setup.

curcuz, 2017-06-06:
thanks @Aubron I really appreciate it. Will keep investigating on my end ofc

joeyvandijk, 2017-06-06:
@curcuz I have tried with the new Resin OS (v2) release 2 weeks ago, but it did not work.

curcuz, 2017-06-06:
@joeyvandijk this seems to be an electron problem, see https://github.com/electron/electron/issues/4677

lurch, 2017-06-06:
@curcuz : @jviotti also recently said that he was going to attempt building an ARMv6 version of electron... ;)
https://gitter.im/resin-io/etcher?at=5935a3e131f589c64f7927bb

steffenmllr, 2017-09-16:
I have question about the zero: Does resin-electronjs work on a raspberry zero ?

willnewton, 2017-09-18:
@steffenmllr I think that should work. The best place to ask questions about using Resin is the forums: https://forums.resin.io/

lurch, 2017-09-18:
I haven't tested myself, but AFAIK Electron (and therefore resin-electronjs) only works on an ARMv7 CPU (i.e. Pi2 or Pi3), and therefore won't work on the PiZero as it only has an ARMv6 CPU. See #52

closed 2018-02-09 by curcuz