balena-io-experimental/balena-electronjs #20: Adding React to to boilerplate

jakkra, 2016-12-14:
I'm trying to integrate ReactJS, but can't get it to work.

Anyone done this before?

danielmahon, 2016-12-14:
Sure. Working on a project using it now. What problems are you having? If it's a routing problem and your using something like react-router, make sure you use "hash history" instead of "browser history" so the urls will work when pointing to your index.html file. Also want to probably like scripts and stuff with relative paths.

jakkra, 2016-12-14:
It's the structure I'm having problems with and to get the Docker file to work. It doesn't want to run. I'm not very good at configuring Web-pack/Bower so some kind of boilerplate would be awesome.

danielmahon, 2017-01-06:
@jakkra Sorry I missed your comment earlier, once I get the performance and usability to where I want it I can work on a boilerplate (towards end of month). In the meantime here is the basic setup I'm using. 

I do NOT claim this to be the optimal way and I might have messed stuff up while pasting and removing private info... 😉 

https://gist.github.com/danielmahon/35f49cc84a9520aec4ef2ed4db7fddd3

danielmahon, 2017-01-10:
FYI: When storing persistent state in react/redux, consider moving the electron/chrome user directory to `/data`. I'm using this to make sure data I save to IndexedDB persists across application restarts.

Dockerfile
```
# Create data directory to store electron config
RUN mkdir -p /data/electron
```

main.js
```javascript
// Set user data directory if RESIN device
// MUST SET BEFORE APP READY
if (process.env.RESIN) {
  app.setPath('userData', '/data/electron');
}

app.on('ready', () => {
...
```

curcuz, 2017-01-10:
@danielmahon great suggestion, I will add this as an option, but will make it disabled by default (we don't want to persist random users data in a kiosk device 🙈  )

danielmahon, 2017-01-10:
@curcuz agreed!

craig-mulligan, 2017-02-05:
For anyone interested, I set up a react boilerplate by mixing resin-electronjs with [react-create-app](https://github.com/facebookincubator/create-react-app). https://github.com/resin-io-playground/resin-electronjs-react. 

@danielmahon are you using node API's in your app?

danielmahon, 2017-02-08:
@craig-mulligan yes. currently my setup is based on https://github.com/chentsulin/electron-react-boilerplate since I am also running a version of the app on a couple surface tablets

imomaliev, 2017-06-15:
For anyone interested in electron-vue integration https://github.com/imomaliev/resin-electron-vue

imomaliev, 2017-08-21:
I think this can be closed

closed 2017-08-21 by curcuz