balena-io-experimental/balena-electronjs #21: Electron failing to start on Intel NUC

shaunmulligan, 2017-01-03:
The following errors were reported:
```
29.12.16 18:22:50 [-0500] [174:1229/232250:ERROR:desktop_window_tree_host_x11.cc(1648)] Not implemented reached in void views::DesktopWindowTreeHostX11::MapWindow(ui::WindowShowState)
29.12.16 18:22:50 [-0500] [194:1229/232250:ERROR:sandbox_linux.cc(343)] InitializeSandbox() called with multiple threads in process gpu-process.
29.12.16 18:22:50 [-0500] [174:1229/232250:ERROR:browser_gpu_channel_host_factory.cc(131)] Failed to create channel.
```

sinback, 2017-01-03:
This problem persisted when I changed package.json to require the latest version of Electron (1.4.13).

curcuz, 2017-01-03:
@sinback does a clean push from `master` produce the same issue (app not starting)? the errors @shaunmulligan reported should not prevent electron to run.

In any case I will give it a spin on NUC soon so I will be hopefully able to reproduce

sinback, 2017-01-03:
@curcuz Today I noticed that npm seems to be having trouble resolving electron's dependency on bluebird, so I can't quite do a successful clean push from master. But after I added bluebird as an explicit dependency I saw the same error. (I didn't change anything else.)

My NUC doesn't seem to be driving the display it's connected to, which makes me assume electron isn't running, or at least isn't running properly. I also don't see any displays listed when I start a terminal in the machine and run xrandr (not sure if this is expected behavior or not since I have no understanding of how X servers work on resinOS). When the NUC ran Ubuntu, it discovered and drove monitors no problem.

Potentially relevant? Some other people seem to have encountered this bug in past electron versions...: https://github.com/electron/electron/issues/3099

curcuz, 2017-01-05:
This is related to recent versions of `electron` and broken `electron-rebuild` - I will drop the `electron-rebuild` dependency in favour of [require-rebuild](https://www.npmjs.com/package/require-rebuild)

expect a fix coming soon in the upcoming days, currently testing locally the fix.

curcuz, 2017-01-05:
@sinback can you please test the `fix-native-modules-rebuild` branch (PR #23 ) via `git push resin fix-native-modules-rebuild:master` ?

sinback, 2017-01-05:
Yep, I pulled down the branch and pushed it to my NUC. problem persists for me.

```05.01.17 14:16:36 [-0500] Systemd init system enabled.
05.01.17 14:16:36 [-0500] systemd 215 running in system mode. (+PAM +AUDIT +SELINUX +IMA +SYSVINIT +LIBCRYPTSETUP +GCRYPT +ACL +XZ -SECCOMP -APPARMOR)
05.01.17 14:16:36 [-0500] Detected virtualization 'other'.
05.01.17 14:16:36 [-0500] Detected architecture 'x86-64'.
05.01.17 14:16:36 [-0500] Set hostname to <intel-corei7-64-7700288>.
05.01.17 14:16:36 [-0500] xauth: file /root/.Xauthority does not exist
05.01.17 14:16:36 [-0500]
05.01.17 14:16:36 [-0500]
05.01.17 14:16:36 [-0500] X.Org X Server 1.16.4
05.01.17 14:16:36 [-0500] Release Date: 2014-12-20
05.01.17 14:16:36 [-0500] X Protocol Version 11, Revision 0
05.01.17 14:16:36 [-0500] Build Operating System: Linux 3.16.0-4-amd64 x86_64 Debian
05.01.17 14:16:36 [-0500] Current Operating System: Linux intel-corei7-64-7700288 4.1.8-yocto-standard #1 SMP PREEMPT Mon Dec 5 15:07:28 UTC 2016 x86_64
05.01.17 14:16:36 [-0500] Kernel command line: BOOT_IMAGE=/vmlinuz root=/dev/sda2 rootwait
05.01.17 14:16:36 [-0500] Build Date: 11 February 2015 12:32:02AM
05.01.17 14:16:36 [-0500] xorg-server 2:1.16.4-1 (http://www.debian.org/support)
05.01.17 14:16:36 [-0500] Current version of pixman: 0.32.6
05.01.17 14:16:36 [-0500] Before reporting problems, check http://wiki.x.org
05.01.17 14:16:36 [-0500] to make sure that you have the latest version.
05.01.17 14:16:36 [-0500] Markers: (--) probed, (**) from config file, (==) default setting,
05.01.17 14:16:36 [-0500] (++) from command line, (!!) notice, (II) informational,
05.01.17 14:16:36 [-0500] (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
05.01.17 14:16:36 [-0500] (==) Log file: "/var/log/Xorg.0.log", Time: Thu Jan 5 19:16:36 2017
05.01.17 14:16:36 [-0500] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
05.01.17 14:16:38 [-0500] [174:0105/191638:WARNING:audio_manager.cc(314)] Multiple instances of AudioManager detected
05.01.17 14:16:38 [-0500] [174:0105/191638:WARNING:audio_manager.cc(275)] Multiple instances of AudioManager detected
05.01.17 14:16:38 [-0500] [174:0105/191638:ERROR:desktop_window_tree_host_x11.cc(1648)] Not implemented reached in void views::DesktopWindowTreeHostX11::MapWindow(ui::WindowShowState)
05.01.17 14:16:38 [-0500] [193:0105/191638:ERROR:sandbox_linux.cc(343)] InitializeSandbox() called with multiple threads in process gpu-process.
05.01.17 14:16:38 [-0500] [174:0105/191638:ERROR:browser_gpu_channel_host_factory.cc(131)] Failed to create channel.```

curcuz, 2017-01-05:
@sinback the logs you pasted are unharmful and should not prevent electron from running - do you see any output on the screen when the device boots? you should at least see the `resin.io - check your dashboard' ascii message

curcuz, 2017-01-07:
merging #23 and closing this since it got confirmed working by others. Feel free to re-open if issue persists nad you can confirm the display has being detected from your device

closed 2017-01-07 by curcuz

mateodelnorte, 2018-04-09:
I'm experiencing this issue. Any info on a fix?

PR #23 (2017-01-07): drop electron-rebuild in favour of require-rebuild
fixes #21 #22