balena-io-experimental/balena-electronjs #22: Cannot find bluebird

sGerli, 2017-01-05:
Hi, I'm getting an error after running `git push resin master`:
I don't know if I'm doing anything wrong, and I haven't modified anything yet. I'm running this from windows 10 and I'm trying to deploy to a Raspberry Pi 2.
Logs:
```
Building on 'local'
-----> Dockerfile template application detected
Step 1 : FROM resin/raspberrypi2-node:slim
 ---> b5a42d379865
Step 2 : RUN apt-get update && apt-get install -y   wget   curl && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> e4624b748864
Step 3 : RUN sed -i "s!httpredir.debian.org!`curl -s -D - http://httpredir.debian.org/demo/debian/ | awk '/^Link:/ { print $2 }' | sed -e 's@<http://\(.*\)/debian/>;@\1@g'`!" /etc/apt/sources.list
 ---> Using cache
 ---> 154dd2adc3d8
Step 4 : RUN apt-get update && apt-get install -y   apt-utils   build-essential   clang   xserver-xorg-core   xserver-xorg-input-all   xserver-xorg-video-fbdev   xorg   libdbus-1-dev   libgtk2.0-dev   libnotify-dev   libgnome-keyring-dev   libgconf2-dev   libasound2-dev   libcap-dev   libcups2-dev   libxtst-dev   libxss1   libnss3-dev   fluxbox   libsmbclient   libssh-4   fbset   libexpat-dev && rm -rf /var/lib/apt/lists/*
 ---> Using cache
 ---> bc9e7a6410a6
Step 5 : RUN mkdir ~/.fluxbox
 ---> Using cache
 ---> 86b4dc24ecb8
Step 6 : RUN echo "xset s off" > ~/.fluxbox/startup && echo "xserver-command=X -s 0 dpms" >> ~/.fluxbox/startup
 ---> Using cache
 ---> 703520f66b58
Step 7 : RUN echo "#!/bin/bash" > /etc/X11/xinit/xserverrc   && echo "" >> /etc/X11/xinit/xserverrc   && echo 'exec /usr/bin/X -s 0 dpms -nocursor -nolisten tcp "$@"' >> /etc/X11/xinit/xserverrc
 ---> Using cache
 ---> ffc6e402d725
Step 8 : RUN npm config set unsafe-perm true
 ---> Using cache
 ---> 8b11d1f32ddb
Step 9 : RUN printf "%s\n" "${PWD##}" > SOURCEFOLDER
 ---> Using cache
 ---> 4d70194d8650
Step 10 : WORKDIR /usr/src/app
 ---> Using cache
 ---> 79014102b5f7
Step 11 : COPY "$SOURCEFOLDER/app/package.json" ./
 ---> Using cache
 ---> e19aaf86c44b
Step 12 : RUN JOBS=MAX npm i --production
 ---> Using cache
 ---> aa342ad2904f
Step 13 : COPY "$SOURCEFOLDER/app" ./
 ---> Using cache
 ---> 13d8ed80f0e0
Step 14 : RUN ./node_modules/.bin/electron-rebuild
 ---> Running in a9f0f0bbb094
module.js:472
    throw err;
    ^

Error: Cannot find module 'bluebird'
    at Function.Module._resolveFilename (module.js:470:15)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/usr/src/app/node_modules/electron-rebuild/lib/cli.js:8:17)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)
Removing intermediate container a9f0f0bbb094
       Error while processing push: Error: The command '/bin/sh -c ./node_modules/.bin/electron-rebuild' returned a non-zero code: 1

remote: error: hook declined to update refs/heads/master
```

curcuz, 2017-01-05:
Hey @sGerli this seems to be the same issue described in #21  

I managed to reproduce, will push a fix very soon after some more local test.

Long story short this is related to recent versions of `electron` and broken `electron-rebuild` - I will drop the `electron-rebuild` dependency in favour of [require-rebuild](https://www.npmjs.com/package/require-rebuild)

sGerli, 2017-01-05:
Ok, thanks

curcuz, 2017-01-05:
@sGerli can you please test the `fix-native-modules-rebuild` branch (PR #23 ) via `git push resin fix-native-modules-rebuild:master` ?

sGerli, 2017-01-05:
Yes this one worked 👍

curcuz, 2017-01-05:
perfect, expect #23 to be merged soon, will keep this open until then

closed 2017-01-07 by curcuz