balena-io-experimental/balena-electronjs #36: adding "pi-pins" to this template

derrickaxel, 2017-02-01:
Hi
I am trying to use the GPIOs so I added "pi-pins". my dependencies in my package.json look like this 
```
"dependencies": {
        "bootstrap": "^4.0.0-alpha.4",
        "electron": "^1.4.13",
        "firebase": "^3.6.8",
        "jquery": "^3.1.0",
        "moment": "^2.17.1",
        "pi-pins": "~1.0.2",
        "require-rebuild": "^1.2.8",
        "tether": "^1.3.7",
        "weather-js": "^1.0.2"
    }
```

when I tried to run it on my RPI3 i got this error message:

> [194:0201/234145:INFO:CONSOLE(83)] "Uncaught Error: Module version mismatch. Expected 50, got 48.", source: /usr/src/app/node_modules/bindings/bindings.js (83)

if I Remove the code below (which is in my index.js) the error goes away:

```
if (isPi) {
        var GPIO = require('pi-pins');
    }
```

I am doing something wrong?

lurch, 2017-02-02:
Yeah, I've seen this error before on [another project](https://github.com/resin-io/etcher/pull/802#issuecomment-257415982) - the problem is that your nodejs runtime uses one ABI version, and the electron runtime uses a different ABI version, so depending on how you install the node modules, they either won't work with nodejs, or they won't work with electron.
I've not actually used the code in this resin-electronjs repo myself yet though, so maybe one of the other devs will be able to offer further pointers.

sGerli, 2017-02-02:
I'm having the same problem https://forums.resin.io/t/pi-pins-not-working-with-electron-project/548 .
That sounds like a good answer but isn't `require-rebuild` there to fix this types of issues?
If you find a solution please let me know.

derrickaxel, 2017-02-02:
@sGerli I have tried `require-rebuild` but it did not change anything...

derrickaxel, 2017-02-02:
@sGerli 
Ok I got it to work by reverting my electron in my package.json to 1.2.5.

sGerli, 2017-02-02:
@derrickaxel  How did you revert to electron 1.2.5? When I tried `"electron": "1.2.5",` I got the following error:
```
npm ERR! Linux 4.7.3-coreos-r3
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "--unsafe-perm" "--production"
npm ERR! node v6.9.2
npm ERR! npm  v3.10.9
npm ERR! code ETARGET

npm ERR! notarget No compatible version found: electron@1.2.5
npm ERR! notarget Valid install targets:
npm ERR! notarget 1.5.0, 1.4.15, 1.4.14, 1.4.13, 1.4.12, 1.4.11, 1.4.10, 1.4.8, 1.4.7, 1.4.6, 1.4.5, 1.4.4, 1.4.3, 1.4.2, 1.4.1, 1.4.0, 1.3.13, 1.3.12, 1.3.10, 1.3.9, 1.3.8, 1.3.7, 1.3.6, 1.3.5, 1.3.4, 1.3.3, 1.3.2, 1.3.1, 0.4.1, 0.4.0, 0.3.0, 0.2.1, 0.2.0, 0.1.2, 0.1.1, 0.1.0
npm ERR! notarget
npm ERR! notarget This is most likely not a problem with npm itself.
npm ERR! notarget In most cases you or one of your dependencies are requesting
npm ERR! notarget a package version that doesn't exist.
npm ERR! notarget
npm ERR! notarget It was specified as a dependency of 'clockos'
npm ERR! notarget

npm ERR! Please include the following file with any support request:
npm ERR!     /usr/src/app/npm-debug.log
Removing intermediate container b22cb931e8cc
Build failed: BuildError: The command '/bin/sh -c JOBS=MAX npm install --unsafe-perm --production       && npm cache clean' returned a non-zero code: 1
```
It isn't finding electron 1.2.5

derrickaxel, 2017-02-02:
@sGerli I will check my setup tonight when I get home and I will let you know.
but my dependencies look like this 
```
"dependencies": {
        "bootstrap": "^4.0.0-alpha.4",
        "electron": "^1.2.5",
        "firebase": "^3.6.8",
        "jquery": "^3.1.0",
        "moment": "^2.17.1",
        "pi-pins": "~1.0.2",
        "require-rebuild": "^1.2.8",
        "tether": "^1.3.7",
        "weather-js": "^1.0.2"
    }
```
Mine compiled fine. are you on a RPI?
here are 2 links to compare the electron ABI version and the node ABI version 
electron: https://www.npmjs.com/package/electron-abi
node : https://nodejs.org/en/download/releases/
you need to get them to match.
maybe you could use this https://www.npmjs.com/package/node-abi
or maybe you could install a new version of nodejs....

sGerli, 2017-02-02:
That's weird, I believe I'm using node 6.9.2 because of this first line in the dockerfile `FROM resin/%%RESIN_MACHINE_NAME%%-node:6.9.2`, but when I try it using electron 1.2.5 with `^` my build chooses 1.4.15 and when I use it without it it says:
```
npm ERR! notarget No compatible version found: electron@1.2.5
npm ERR! notarget Valid install targets:
npm ERR! notarget 1.5.0, 1.4.15, 1.4.14, 1.4.13, 1.4.12, 1.4.11, 1.4.10, 1.4.8, 1.4.7, 1.4.6, 1.4.5, 1.4.4, 1.4.3, 1.4.2, 1.4.1, 1.4.0, 1.3.13, 1.3.12, 1.3.10, 1.3.9, 1.3.8, 1.3.7, 1.3.6, 1.3.5, 1.3.4, 1.3.3, 1.3.2, 1.3.1, 0.4.1, 0.4.0, 0.3.0, 0.2.1, 0.2.0, 0.1.2, 0.1.1, 0.1.0
```

derrickaxel, 2017-02-02:
ok when I get home I will double check everything and I will let you know

curcuz, 2017-02-02:
reverting to a very old version of electron does not seem a proper fix to me. Reopening this

curcuz, 2017-02-02:
@derrickaxel we should not add `pi-pins` to this template since that's business-logic specific (a user might want to use a different module) but we should ensure it works ;)

derrickaxel, 2017-02-02:
@sGerli  you are right i got excited an forgot to uncomment the line with `var GPIO = require('pi-pins');` to do my test. sorry....   
but it looks like @curcuz has a fix for us

jviotti, 2017-02-03:
@curcuz The fix is to configure npm correctly to build against the Electron v8 headers instead of the local ones. See https://github.com/resin-io/etcher/blob/master/scripts/build/dependencies-npm.sh#L82. There are other environment variables you have to set. Here's a full example:

```sh
export npm_config_disturl=https://atom.io/download/electron
export npm_config_runtime=electron
export npm_config_target=<electron version this project is using, e.g: 1.4.4>
export npm_config_build_from_source=true # we need to ensure we build from source so the settings above get applied
export npm_config_arch=<architecture>
```

I recommend you to check our bash script to install npm dependencies, and probably even re-use it here: https://github.com/resin-io/etcher/blob/master/scripts/build/dependencies-npm.sh#L82.

We plan to release all those build scripts as a standalone project, so both resin-electronjs and Etcher can benefit from it.

derrickaxel, 2017-02-04:
@sGerli I got it to pass the compile stage by doing this:
in the dockerfile  `FROM resin/raspberrypi3-node:7.4.0`
in the package.json `"electron": "^1.5.0",`

but I am not sure I fixed the issue because now I am facing another challenge which is :
"Uncaught TypeError: Cannot read property 'value' of undefined", source: file:///usr/src/app/data/src/index.js (47)

```
this might be a permission issue but again I am not sure...

if (isPi) {
        var GPIO = require('pi-pins');
        var pin = GPIO.connect(13).mode('out');
        pin.value(true);
    }

```

sGerli, 2017-02-04:
@derrickaxel to fix mine I applied the fix provided by #37 and now it's working perfectly.

derrickaxel, 2017-02-04:
@sGerli got it... I thought it needed to be compiled for some reason  thanks it will do it too

PR #37 (2017-02-10): switch (again) to electron-rebuild
fixes #36 

this allows free usage of latest `electron` versions and moves finally back to the builder the native module rebuilding
closed 2017-02-10 by curcuz