balena-io-experimental/balena-electronjs #44: Large image handling crash

OllieLy, 2017-06-25:
If I use any large images or do anything a little challenging within the renderer process chromium crashes not giving any error codes etc. The screen basically just goes blank.

I am using an Intel NUC7I5BNK with a pcap screen

Here is a fork of this repo that just has a large image set as a background, if you try to zoom in to it you will get the same issue that I have here. - https://github.com/OllieLy/resin-electronjs

Every now and again I get this on the resin logs:

```
[182:0625/154212.641834:FATAL:bus.cc(1197)] D-Bus connection was disconnected. Aborting.
#0 0x000001cd013e <unknown>
#1 0x000001cb53db <unknown>
#2 0x000002fe74b0 <unknown>
#3 0x7f3a25eea1d1 dbus_connection_dispatch
#4 0x000002fe8109 <unknown>
#5 0x000001cb883e <unknown>
#6 0x000001d0b05c <unknown>
#7 0x000001d0b308 <unknown>
#8 0x000001d0b66b <unknown>
#9 0x000001cbab99 <unknown>
#10 0x000001d0adf7 <unknown>
#11 0x000001ced600 <unknown>
#12 0x000001d22fb5 <unknown>
#13 0x000001cfd753 <unknown>
#14 0x7f3a2809c064 start_thread
#15 0x7f3a214ce62d clone

xinit: connection to X server lost

waiting for X server to shut down (EE) Server terminated successfully (0). Closing log file
```

lekkas, 2017-06-27:
Hey @OllieLy , thanks for the report. @craig-mulligan or @curcuz may have seen this error before.

imrehg, 2017-06-28:
I've checked this on a NUC and on a RPi3 given the linked repo, and both of them crash with white-blank screen on pinch-zoom.

I didn't have any logs for this anywhere (neither the container nor the host OS, so couldn't see something like mentioned above). Bringing up the debug console, on the zoom crash the debug console disconnects too, so no log there either...

PR #45 (2017-07-05): increase available shm memory size
fixes #44
commit ac171979 (2017-07-05): increase available shm memory size

fixes #44
commit 76e348f2 (2017-07-27): increase available shm memory size

fixes #44
closed 2017-07-05 by petrosagg