balena-io-experimental/balena-electronjs #48: No address associated with hostname

arunesh90, 2017-07-18:
I'm not able to launch the application. I've followed the instructions exactly as explained on the repo but when I boot the pi up, it doesn't seem to launch. This is on a raspberry pi zero.
Log:
```
18.07.17 05:12:16 (+0200) hostname: No address associated with hostname
18.07.17 05:12:16 (+0200) xauth:  file /root/.Xauthority does not exist
18.07.17 05:12:16 (+0200) xauth: (stdin):1:  bad display name "eab746c:0" in "add" command
18.07.17 05:12:16 (+0200) 
18.07.17 05:12:16 (+0200) 
18.07.17 05:12:16 (+0200) X.Org X Server 1.18.4
18.07.17 05:12:16 (+0200) Release Date: 2016-07-19
18.07.17 05:12:16 (+0200) X Protocol Version 11, Revision 0
18.07.17 05:12:16 (+0200) Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
18.07.17 05:12:16 (+0200) Current Operating System: Linux eab746c 4.4.50 #2 Tue Jul 4 20:41:20 UTC 2017 armv6l
18.07.17 05:12:16 (+0200) Kernel command line: bcm2708_fb.fbwidth=1280 bcm2708_fb.fbheight=1024 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 dma.dmachans=0x7f35 bcm2708.boardrev=0x900093 bcm2708.serial=0x4b88cbc6 bcm2708.uart_clock=48000000 bcm2708.disk_led_gpio=47 smsc95xx.macaddr=B8:27:EB:88:CB:C6 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
18.07.17 05:12:16 (+0200) Build Date: 11 November 2016  11:59:59AM
18.07.17 05:12:16 (+0200) xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support) 
18.07.17 05:12:16 (+0200) Current version of pixman: 0.33.3
18.07.17 05:12:16 (+0200) 	Before reporting problems, check http://wiki.x.org
18.07.17 05:12:16 (+0200) 	to make sure that you have the latest version.
18.07.17 05:12:16 (+0200) Markers: (--) probed, (**) from config file, (==) default setting,
18.07.17 05:12:16 (+0200) 	(++) from command line, (!!) notice, (II) informational,
18.07.17 05:12:16 (+0200) 	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
18.07.17 05:12:16 (+0200) (==) Log file: "/var/log/Xorg.0.log", Time: Tue Jul 18 03:12:16 2017
18.07.17 05:12:16 (+0200) (==) Using system config directory "/usr/share/X11/xorg.conf.d"
18.07.17 05:12:22 (+0200) xinit: connection to X server lost
18.07.17 05:12:22 (+0200) 

18.07.17 05:12:22 (+0200) waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
18.07.17 05:12:23 (+0200) 
18.07.17 05:12:23 (+0200) xauth: (argv):1:  bad display name "eab746c:0" in "remove" command
```

curcuz, 2017-08-17:
Hi @arunesh90 this is due to electron being broken on armv6 https://github.com/electron/electron/issues/4677

curcuz, 2017-08-21:
related to #52

closed 2017-08-21 by curcuz