balena-io-experimental/balena-electronjs #50: mention there is a "intel" branch that is meant to be a reference to enable full HW acceleration on Intel devices

curcuz, 2017-08-02:


kristjanmik, 2017-10-02:
Hm, this link is not acceptable. Could you link the discussion here as well?

curcuz, 2017-10-02:
hey @kristjanmik this repository is optimised for Intel chipsets and grants hw acceleration https://github.com/resin-io/resin-electronjs/tree/intel

concreted, 2017-12-19:
@curcuz Are there details posted somewhere on what hardware the Intel branch is meant to work with? My team tried using the Dockerfile from that branch on two Intel Compute Stick models, and it improved performance on our Atom stick but not our Core M3 stick.

curcuz, 2018-01-03:
@concreted it is targeting the Intel NUC family and UPboard family (or at least has been tested against only those) - good point in writing it down somewhere in the README