balena-io-experimental/balena-electronjs #52: /lib/arm-linux-gnueabihf/libdl.so.2: invalid ELF header

mexpolk, 2017-08-20:
For some reason my app crashes with the following error:

```
20.08.17 01:08:04 (-0500) Restarting application 'registry2.resin.io/classroompoc/efce98b5c08cb6929b3940d2d4f889381dabf982'
20.08.17 01:04:09 (-0500) /bin/bash: error while loading shared libraries: /lib/arm-linux-gnueabihf/libdl.so.2: invalid ELF header
20.08.17 01:05:00 (-0500) /bin/bash: error while loading shared libraries: /lib/arm-linux-gnueabihf/libdl.so.2: invalid ELF header
20.08.17 01:05:49 (-0500) /bin/bash: error while loading shared libraries: /lib/arm-linux-gnueabihf/libdl.so.2: invalid ELF header
20.08.17 01:06:35 (-0500) /bin/bash: error while loading shared libraries: /lib/arm-linux-gnueabihf/libdl.so.2: invalid ELF header
20.08.17 01:07:20 (-0500) /bin/bash: error while loading shared libraries: /lib/arm-linux-gnueabihf/libdl.so.2: invalid ELF header
20.08.17 01:08:04 (-0500) /bin/bash: error while loading shared libraries: /lib/arm-linux-gnueabihf/libdl.so.2: invalid ELF header
20.08.17 01:08:07 (-0500) Application exited 'registry2.resin.io/classroompoc/efce98b5c08cb6929b3940d2d4f889381dab
```

Also, I'm unable to enter into the terminal to dig more on the issue and check if it is a permission problem.

I'm using the example provided in https://github.com/resin-io/resin-electronjs, with `Raspberry Pi v1 B+`. Any clues about what I may be missing? Is this a known problem?

Thanks!

lurch, 2017-08-20:
Electron only runs on ARMv7 (or above) CPUs, while the Raspberry Pi B+ only has an ARMv6 CPU.
I.e. if you want to run Electron, you'll need to upgrade to either a Raspberry Pi 2 or a Raspberry Pi 3.

closed 2017-08-21 by curcuz

jollytoad, 2017-09-04:
It would be really useful if this was mentioned in the README, which currently gives the impression that it'll be ok on a Pi 1.

lurch, 2017-09-04:
ping @curcuz

dazz100, 2018-04-16:
I have read the above explanation and I am trying to install mjpg-streamer on a Pi 3B following the instructions here:  https://github.com/jacksonliam/mjpg-streamer
I got the error:

    /usr/lib/arm-linux-gnueabihf/libbfd-2.28-system.so: invalid ELF header
    CMakeFiles/cmTC_9bf69.dir/build.make:65: recipe for target
    'CMakeFiles/cmTC_9bf69.dir/testCCompiler.c.o' failed

 Is there a fix?

curcuz, 2018-04-16:
@dazz100 can you please elaborate on how your comment is related to the issue or the project in general?

dazz100, 2018-04-16:
Hi

I think the solution to my problem is here:

https://github.com/bytedeco/javacpp-presets/issues/486

Too late to try that now.    Will give it ago tomorrow.

/by vb216/

/Ah I see. I need to break out the wiki instructions for sure then, a 
section for native install which doesn't need any of that stuff. So it 
sounds like you're right, the update-alternatives will have moved the 
links away from what we'd like. Could you send me the output from this://
//ls -tlra armgcc/

//

/I guess it has entries like this?//
//arm-linux-gnueabihf-gcc -> /etc/alternatives/arm-linux-gnueabihf-gcc/

//

/Unfortunately, I don't use update-alternatives so much, so I'm not sure 
quickly how to clean it up. To me, I think easier option would be to 
remove this symlink to /etc/alternatives/arm-linux-gnueabihf-gcc, and 
create a new one pointing to gcc. In theory, you could always put the 
old symlink back. If you have time, it might not be a bad idea to take 
the SD card out and make a quick backup./

//

/But, I recreated the steps on a Pi here, and this seems to fix it for me:/

//

/Before test://
//cd /usr/bin//
//./arm-linux-gnueabihf-gcc//
//-bash: ./arm-linux-gnueabihf-gcc: cannot execute binary file: Exec 
format error//
//(because its trying a x86 build of the compiler so can't execute it)/

//

/Fix://
//sudo rm arm-linux-gnueabihf-gcc//
//sudo ln -s gcc arm-linux-gnueabihf-gcc/

//

/After fix://
//./arm-linux-gnueabihf-gcc//
//arm-linux-gnueabihf-gcc: fatal error: no input files/

//

/You'll need to do the same for arm-linux-gnueabihf-g++ and 
arm-linux-gnueabihf-cpp/

vb216 refers to a Wiki without saying where that Wiki is.

Regards

Darren Conway
36 Orr Crescent
Lower Hutt
New Zealand
ph +64  (0)4 569 1963

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

curcuz, 2018-04-16:
@dazz100 sorry I don't understand - this thread is about electron and rpi0/1, while you are talking about installing mjpg-streamer on a rpi3 :P

lurch, 2018-04-16:
@dazz100 The issue you're commenting on here is entirely unrelated to your problem. You'd be much better off asking for help at https://github.com/jacksonliam/mjpg-streamer#discussion--questions--help

dazz100, 2018-04-16:
Hi

You are right.  Sorry.

Thanks for the link

Regards

Darren Conway
36 Orr Crescent
Lower Hutt
New Zealand
ph +64  (0)4 569 1963

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus