balena-io-experimental/balena-electronjs #53: Node not running

WoodyPhil, 2017-09-24:
When i put the electron files into the app directory it fire´s up the index.html but give me errors like 

[image]

In my opinion the node server is not running, but i have checked the environment variable

[image]

I´m stuck with that for a long time some help would be really nice

curcuz, 2017-11-15:
hi @WoodyPhil the errors you showed are not related with node running or not - in fact, it **is** running and that's the cause of your issue. Please have a read here about jQuery being confused on electron https://stackoverflow.com/questions/32621988/electron-jquery-is-not-defined

curcuz, 2018-01-03:
closing since the user didn't reply in a long time

closed 2018-01-03 by curcuz