balena-io-experimental/balena-electronjs #55: Can't use kiosk mode

maximilliangeorge, 2017-11-14:
Kiosk mode prevents 'ready-to-show' from firing. Also, it doesn't hide the top menu. I've tried modifying fluxbox settings with no luck... Any hints?

curcuz, 2017-11-15:
Hi @maximilliangeorge kiosk mode has nothing to do with Fluxbox - it is just one of the many browserWindow parameters that I wrap with the env vars mechanism https://electron.atom.io/docs/api/browser-window/#new-browserwindowoptions

curcuz, 2017-11-15:
so I guess the focus of the investigation would be on electron itself, or on how we integrate it into this project

curcuz, 2018-02-09:
since the user didn't reply for a long time, I'm going to assume he fixed his problem and it wasn't related to this repository.

closed 2018-02-09 by curcuz