balena-io-experimental/balena-electronjs #57: RPi startx "bad display name"

benjsec, 2017-12-18:
I'm trying to run this on a raspberry pi B and the display (startx) is failing. The log output is

```
18.12.17 08:52:47 (+0000) Systemd init system enabled.
18.12.17 08:52:48 (+0000) systemd 215 running in system mode. (+PAM +AUDIT +SELINUX+IMA +SYSVINIT +LIBCRYPTSETUP +GCRYPT +ACL +XZ -SECCOMP -APPARMOR)
18.12.17 08:52:48 (+0000) Detected virtualization 'other'.
18.12.17 08:52:48 (+0000) Detected architecture 'arm'.
18.12.17 08:52:48 (+0000) Set hostname to <f9dcb8d>.
18.12.17 08:53:27 (+0000) hostname: No address associated with hostname
18.12.17 08:53:27 (+0000) xauth:  file /root/.Xauthority does not exist
18.12.17 08:53:27 (+0000) xauth: (stdin):1:  bad display name "f9dcb8d:0" in "add" command
18.12.17 08:53:27 (+0000)
18.12.17 08:53:28 (+0000)
18.12.17 08:53:28 (+0000) X.Org X Server 1.18.4
18.12.17 08:53:28 (+0000) Release Date: 2016-07-19
18.12.17 08:53:28 (+0000) X Protocol Version 11, Revision 0
18.12.17 08:53:28 (+0000) Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
18.12.17 08:53:28 (+0000) Current Operating System: Linux f9dcb8d 4.4.50 #2 Mon Jul31 10:26:25 UTC 2017 armv6l
18.12.17 08:53:28 (+0000) Kernel command line: bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 dma.dmachans=0x7f35 bcm2708.boardrev=0xe bcm2708.serial=0x6608a896 bcm2708.uart_clock=48000000 smsc95xx.macaddr=B8:27:EB:08:A8:96 vc_mem.mem_base=0x1fa00000 vc_mem.mem_size=0x20000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyAMA0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
18.12.17 08:53:28 (+0000) Build Date: 11 November 2016  11:59:59AM
18.12.17 08:53:28 (+0000) xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
18.12.17 08:53:28 (+0000) Current version of pixman: 0.33.3
18.12.17 08:53:28 (+0000)       Before reporting problems, check http://wiki.x.org
18.12.17 08:53:28 (+0000)       to make sure that you have the latest version.
18.12.17 08:53:28 (+0000) Markers: (--) probed, (**) from config file, (==) defaultsetting,
18.12.17 08:53:28 (+0000)       (++) from command line, (!!) notice, (II) informational,
18.12.17 08:53:28 (+0000)       (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
18.12.17 08:53:28 (+0000) (==) Log file: "/var/log/Xorg.0.log", Time: Mon Dec 18 08:53:28 2017
18.12.17 08:53:28 (+0000) (==) Using system config directory "/usr/share/X11/xorg.conf.d"
18.12.17 08:53:35 (+0000) xinit: connection to X server lost
18.12.17 08:53:35 (+0000)
18.12.17 08:53:36 (+0000) waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.
18.12.17 08:53:36 (+0000)
18.12.17 08:53:39 (+0000) xauth: (argv):1:  bad display name "f9dcb8d:0" in "remove" command
```

I've tried a couple of different builds, including switching from `-node:6` to `-node:latest` but all seem to do the same. Trying to manually run `startx` either directly on the host or via the resin control panel doens't work either.

curcuz, 2018-01-03:
@benjsec afaik the rpi model B is armv6, which is not currently supported by electron: https://github.com/electron/electron/issues/4677

benjsec, 2018-01-04:
@curcuz thank you for the reply. I've seen the issues with electron on ARM6, and was expecting to have to tackle that at some stage. The error I'm getting here is starting the xserver to bring up the fluxbox desktop. I was expecting this to work, even if electron then didn't, but might have misunderstood how closely linked the packages are.

thegrandoverseer, 2018-02-08:
I am getting the same error on RPI 3 on armv7l architecture.

`08.02.18 19:52:39 (+0000) X.Org X Server 1.18.4
08.02.18 19:52:39 (+0000) Release Date: 2016-07-19
08.02.18 19:52:39 (+0000) X Protocol Version 11, Revision 0
08.02.18 19:52:39 (+0000) Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
08.02.18 19:52:39 (+0000) Current Operating System: Linux d8f2517 4.9.50 #2 SMP Fri Nov 17 17:55:45 UTC 2017 armv7l
08.02.18 19:52:39 (+0000) Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
08.02.18 19:52:39 (+0000) Build Date: 11 November 2016  11:59:59AM
08.02.18 19:52:39 (+0000) xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support) 
08.02.18 19:52:39 (+0000) Current version of pixman: 0.33.3
08.02.18 19:52:39 (+0000) 	Before reporting problems, check http://wiki.x.org
08.02.18 19:52:39 (+0000) 	to make sure that you have the latest version.
08.02.18 19:52:39 (+0000) Markers: (--) probed, (**) from config file, (==) default setting,
08.02.18 19:52:39 (+0000) 	(++) from command line, (!!) notice, (II) informational,
08.02.18 19:52:39 (+0000) 	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
08.02.18 19:52:39 (+0000) (==) Log file: "/var/log/Xorg.0.log", Time: Thu Feb  8 19:52:39 2018
08.02.18 19:52:39 (+0000) (==) Using system config directory "/usr/share/X11/xorg.conf.d"
08.02.18 19:52:39 (+0000) (EE) 
08.02.18 19:52:39 (+0000) (EE) Backtrace:
08.02.18 19:52:39 (+0000) (EE) 
08.02.18 19:52:39 (+0000) (EE) Segmentation fault at address 0x0
08.02.18 19:52:39 (+0000) (EE) 
08.02.18 19:52:39 (+0000) Fatal server error:
08.02.18 19:52:39 (+0000) (EE) Caught signal 11 (Segmentation fault). Server aborting
08.02.18 19:52:39 (+0000) (EE) 
08.02.18 19:52:39 (+0000) (EE) 
08.02.18 19:52:39 (+0000) Please consult the The X.Org Foundation support 
08.02.18 19:52:39 (+0000) 	 at http://wiki.x.org
08.02.18 19:52:39 (+0000)  for help. 
08.02.18 19:52:39 (+0000) (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
08.02.18 19:52:39 (+0000) (EE) 
08.02.18 19:52:39 (+0000) (EE) Server terminated with error (1). Closing log file.
08.02.18 19:52:54 (+0000) xinit: giving up
08.02.18 19:52:54 (+0000) xinit: unable to connect to X server: Connection refused
08.02.18 19:52:54 (+0000) xinit: server error
08.02.18 19:52:54 (+0000) xauth: (argv):1:  bad display name "d8f2517:0" in "remove" command
08.02.18 19:52:54 (+0000) hostname: No address associated with hostname
08.02.18 19:52:54 (+0000) xauth: (stdin):1:  bad display name "d8f2517:0" in "add" command`

adipascu, 2018-02-09:
My app is running corectly but I get a few warnings on each startup:
```
09.02.18 10:39:38 (+0200) Systemd init system enabled.
09.02.18 10:39:38 (+0200) systemd 215 running in system mode. (+PAM +AUDIT +SELINUX+IMA +SYSVINIT +LIBCRYPTSETUP +GCRYPT +ACL +XZ -SECCOMP -APPARMOR)
09.02.18 10:39:38 (+0200) Detected virtualization 'other'.
09.02.18 10:39:38 (+0200) Detected architecture 'arm'.
09.02.18 10:39:38 (+0200) Set hostname to <326338f>.
09.02.18 10:39:40 (+0200) hostname: No address associated with hostname
09.02.18 10:39:40 (+0200) xauth:  file /root/.Xauthority does not exist
09.02.18 10:39:40 (+0200) xauth: (stdin):1:  bad display name "326338f:0" in "add" command
09.02.18 10:39:40 (+0200)
09.02.18 10:39:41 (+0200)
09.02.18 10:39:41 (+0200) X.Org X Server 1.18.4
09.02.18 10:39:41 (+0200) Release Date: 2016-07-19
09.02.18 10:39:41 (+0200) X Protocol Version 11, Revision 0
09.02.18 10:39:41 (+0200) Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
09.02.18 10:39:41 (+0200) Current Operating System: Linux 326338f 4.9.50 #2 SMP WedNov 1 16:03:39 UTC 2017 armv7l
09.02.18 10:39:41 (+0200) Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1366 bcm2708_fb.fbheight=768 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=null root=/dev/mmcblk0p2 rootfstype=ext4 rootwait vt.global_cursor_default=0
09.02.18 10:39:41 (+0200) Build Date: 11 November 2016  11:59:59AM
09.02.18 10:39:41 (+0200) xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
09.02.18 10:39:41 (+0200) Current version of pixman: 0.33.3
09.02.18 10:39:41 (+0200)       Before reporting problems, check http://wiki.x.org
09.02.18 10:39:41 (+0200)       to make sure that you have the latest version.
09.02.18 10:39:41 (+0200) Markers: (--) probed, (**) from config file, (==) defaultsetting,
09.02.18 10:39:41 (+0200)       (++) from command line, (!!) notice, (II) informational,
09.02.18 10:39:41 (+0200)       (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
09.02.18 10:39:41 (+0200) (==) Log file: "/var/log/Xorg.0.log", Time: Fri Feb  9 08:39:41 2018
09.02.18 10:39:41 (+0200) (==) Using system config directory "/usr/share/X11/xorg.conf.d"
09.02.18 10:39:47 (+0200) [209:0209/083947.243195:WARNING:audio_manager.cc(295)] Multiple instances of AudioManager detected
09.02.18 10:39:47 (+0200) [209:0209/083947.243514:WARNING:audio_manager.cc(254)] Multiple instances of AudioManager detected
```
After the last line I can see the runtime log of the newly started app (stdout).

curcuz, 2018-02-09:
@adipascu that is the expected behaviour

benjsec, 2018-02-10:
I did manage to get my hands on an RPi3 and to test with, and the display started fine. There were still some error messages, but they evidently weren't so important.

From what I understand fluxbox _should_ still start on ARM6, even if Electron doesn't work. I did spot in the logs that the Resin build servers are ARM7, so wonder if that could be related? I'd like to be able to dig into this more, but don't really have the time at the moment.

Maybe add in the README that ARM6 (RPi1 and RPi Zero) isn't supported and close this?

PR #65 (2018-02-12): add a warning about armv6 not being supported
fixes #57
closed 2018-02-12 by curcuz