balena-io-experimental/balena-electronjs #58: Writing to DOM blocking other writes?

CodyBranch, 2017-12-29:
I've been working with this project this week. My initial results were great and I was super excited about what could be done with the Pi and Resin/Electron.

While trying to build a more complex page to display with Electron I noticed something odd. I have one socket.io connection that gets data and updates a div 10 times per second, and I have a socket.io connection that gets data and updates a different div every few seconds.

When the every few seconds div gets data and writes to it's div, the 10 times per second div had a hiccup for roughly 3 tenths of a second.

This isn't a problem running on chrome on my laptops browser, or even running and electron app on my windows laptop. Only with the Pi/Resin/Electron combo.

Anyways, I figured I'd throw it out there and see if any of you had a bright idea. I've tried a handful of things to speed up writing the data to the div etc but I still have the same result.

I did end up getting it to run smoothly with a epiphany kiosk option, but I really wanted to stick with Electron since I use it on the windows side of things.

Thanks!

CodyBranch, 2017-12-29:
I event thought maybe I'd try two separate BrowserWindow and it still seemed to act the same as before.

curcuz, 2018-01-03:
Hi @CodyBranch sorry for the late reply - can you share a simple test case app or a snippet I can refer to for debugging?

CodyBranch, 2018-01-08:
Hi @curcuz sorry I didn't get back to you sooner. I've been trying to narrow the problem down in some spare time but was busy last week.

After messing around with it a bit it seems that the hiccup in display only occurs when there is CSS on the page, not just displaying raw data. Could it just be something about a Raspberry Pi's limited power?

When I run a script like this both field populate smoothly, which would indicate it isn't actually the writes to the divs causing the problems, but the render of the CSS itself.

`<script>
    var i = 0;
    setInterval(() => document.getElementById('text').innerHTML = i++, 100);

    var x = 0;
    setInterval(() => document.getElementById('text2').innerHTML = x++ +' - Miusov, as a man man of breeding and deilcacy, could not but feel some inwrd qualms, when he reached the Father Superiors with Ivan: he felt ashamed of havin lost his temper. He felt that he ought to have disdaimed that despicable wretch, Fyodor Pavlovitch, too much to have been upset by him in Father Zossimas cell, and so to have forgotten himself. "Teh monks were not to blame, in any case," he reflceted, on the steps. "And if theyre decent people here (and the Father Superior, I understand, is a nobleman) why not be friendly and courteous withthem? I wont argue, Ill fall in with everything, Ill win them by politness, and show them that Ive nothing to do with that Aesop, thta buffoon, that Pierrot, and have merely been takken in over this affair, just as they have.', 1000);    
  </script>

<body style="background-color:#FFFFFF;">

      <div id="text"></div>
      <div id="text2"></div>
</body>

`

curcuz, 2018-02-09:
@CodyBranch it can be very likely due to the limited CPU resources, since this project is not hw accelerated on the rpi (yet)

CodyBranch, 2018-02-09:
It seems like you're correct. After some more testing I noticed when doing the same things on a small portion of the screen it was fine. When I tried to use the full screen it had issues.

Hopefully someone far smarter than me eventually adds in some HW acceleration :)

curcuz, 2018-02-09:
@CodyBranch I dream about that too :)

Please feel free to close the issue if you agree this is a X11/Chromium limitation on the target architecture

CodyBranch, 2018-02-22:
I agree! Issue closed.

closed 2018-02-22 by CodyBranch