balena-io-experimental/balena-electronjs #60: Chromium crash

ruudboon, 2018-01-13:
I'm running a test on a RaspberryPI 3. It runs fine in the beginning but after a while I get the following error: 

```
../../third_party/tcmalloc/chromium/src/free_list.h:118] Memory corruption detected. 
[270:0113/130440.122161:ERROR:sandbox_linux.cc(343)] InitializeSandbox() called with multiple threads in process gpu-process. 
[270:0113/130441.464521:ERROR:gles2_cmd_decoder.cc(17403)] [.DisplayCompositor-0x50e0d80]GL ERROR :GL_INVALID_OPERATION : glCreateAndConsumeTextureCHROMIUM: invalid mailbox name
[270:0113/130441.620707:ERROR:gles2_cmd_decoder.cc(9636)] [.DisplayCompositor-0x50e0d80]RENDER WARNING: texture bound to texture unit 0 is not renderable. It maybe non-power-of-2 and have incompatible texture filtering.
```

lurch, 2018-01-15:
This project is still using a fairly old version of electron: compare https://github.com/resin-io/resin-electronjs/blob/master/app/package.json#L22 to https://electronjs.org/
@curcuz I wonder if an updated electron version might help here? Or perhaps it'd break more things than it fixes, given that the latest versions of Electron are now using NodeJS version 7?

ruudboon, 2018-01-17:
@lurch I could give it a try this weekend and submit a pull request if it solves the issue.

curcuz, 2018-02-09:
@ruudboon @lurch 

I think it has nothing to do with the electron version - never seen this before

```
../../third_party/tcmalloc/chromium/src/free_list.h:118] Memory corruption detected. 
```