balena-io-experimental/balena-electronjs #62: Doesn

Julio-Guerra, 2018-02-09:
Hi,

I deployed electron on my Raspberry Pi 3 model B as documented and it complains
```
09.02.18 18:15:04 (+0100) Systemd init system enabled.
09.02.18 18:15:04 (+0100) systemd 215 running in system mode. (+PAM +AUDIT +SELINUX +IMA +SYSVINIT +LIBCRYPTSETUP +GCRYPT +ACL +XZ -SECCOMP -APPARMOR)
09.02.18 18:15:04 (+0100) Detected virtualization 'other'.
09.02.18 18:15:04 (+0100) Detected architecture 'arm'.
09.02.18 18:15:04 (+0100) Set hostname to <00aed18>.
09.02.18 18:15:09 (+0100) hostname: No address associated with hostname
09.02.18 18:15:09 (+0100) xauth:  file /root/.Xauthority does not exist
09.02.18 18:15:09 (+0100) xauth: (stdin):1:  bad display name "00aed18:0" in "add" command
09.02.18 18:15:09 (+0100)
09.02.18 18:15:09 (+0100)
09.02.18 18:15:09 (+0100) X.Org X Server 1.18.4
09.02.18 18:15:09 (+0100) Release Date: 2016-07-19
09.02.18 18:15:09 (+0100) X Protocol Version 11, Revision 0
09.02.18 18:15:09 (+0100) Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
09.02.18 18:15:09 (+0100) Current Operating System: Linux 00aed18 4.9.59#2 SMP Wed Feb 7 09:54:42 UTC 2018 armv7l
09.02.18 18:15:09 (+0100) Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1200 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=null root=/dev/mmcblk0p2 rootfstype=ext4 rootwait vt.global_cursor_default=0
09.02.18 18:15:09 (+0100) Build Date: 11 November 2016  11:59:59AM
09.02.18 18:15:09 (+0100) xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
09.02.18 18:15:09 (+0100) Current version of pixman: 0.33.3
09.02.18 18:15:09 (+0100)       Before reporting problems, check http://wiki.x.org
09.02.18 18:15:09 (+0100)       to make sure that you have the latest version.
09.02.18 18:15:09 (+0100) Markers: (--) probed, (**) from config file, (==) default setting,
09.02.18 18:15:09 (+0100)       (++) from command line, (!!) notice, (II) informational,
09.02.18 18:15:09 (+0100)       (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
09.02.18 18:15:09 (+0100) (==) Log file: "/var/log/Xorg.0.log", Time: Fri Feb  9 17:15:09 2018
09.02.18 18:15:09 (+0100) (==) Using system config directory "/usr/share/X11/xorg.conf.d"
09.02.18 18:15:17 (+0100) [202:0209/171517.471452:WARNING:audio_manager.cc(293)] Multiple instances of AudioManager detected
09.02.18 18:15:17 (+0100) [202:0209/171517.471950:WARNING:audio_manager.cc(254)] Multiple instances of AudioManager detected
09.02.18 18:15:18 (+0100) [226:0209/171518.346774:ERROR:sandbox_linux.cc(343)] InitializeSandbox() called with multiple threads in process gpu-process.
09.02.18 18:15:18 (+0100) [202:0209/171518.364568:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
09.02.18 18:15:18 (+0100) [202:0209/171518.364730:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
09.02.18 18:15:18 (+0100) [202:0209/171518.365437:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
09.02.18 18:15:18 (+0100) [202:0209/171518.365698:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
09.02.18 18:15:18 (+0100) [202:0209/171518.366202:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
```

curcuz, 2018-02-09:
I fixed all the errors in https://github.com/resin-io/resin-electronjs/tree/node8

**but** the status is still a blank screen, without any error reported (sigh)

curcuz, 2018-02-09:
I managed to identify the electron binary as the cause of the issue. No matter how I install it, it will always print a blank page and sit there. Will try to get some more logging

curcuz, 2018-02-09:
this is the same issue as https://github.com/electron/electron-quick-start/issues/172 - targeting electron@1.6.15 works

closed 2018-02-09 by curcuz

Julio-Guerra, 2018-02-10:
Thanks a lot for this fast fix. Works perfectly now.

PR #64 (2018-02-09): Node8
fixes #62 #12