balena-io-experimental/balena-electronjs #69: ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.

brownjohnf, 2018-04-09:
It's unclear why this is happening. This is on a push of the master branch as of today's date. Host's config.txt:

```
disable_splash=1
dtparam=i2c_arm=on
dtparam=spi=on
dtparam=audio=on
enable_uart=1
gpu_mem=160
```

brownjohnf, 2018-04-09:
Forgot to say that this is an rpi3 model B, running on an HDMI display.

curcuz, 2018-04-09:
@brownjohnf can you paste more logs around the error please?

curcuz, 2018-04-09:
also, what's the output on the screen after you see this error?

brownjohnf, 2018-04-09:
@curcuz Output on the screen is just white/light gray. Prior to the startup/error, the screen is the default resin display with the ASCII Resin.io on the screen.

```
27.02.18 22:08:50 (-0800)  main  Systemd init system enabled.
27.02.18 22:08:50 (-0800)  main  systemd 215 running in system mode. (+PAM +AUDIT +SELINUX +IMA +SYSVINIT +LIBCRYPTSETUP +GCRYPT +ACL +XZ -SECCOMP -APPARMOR)
27.02.18 22:08:50 (-0800)  main  Detected virtualization 'other'.
27.02.18 22:08:50 (-0800)  main  Detected architecture 'arm'.
27.02.18 22:08:50 (-0800)  main  Set hostname to <cb0459c-cb0459c>.
09.04.18 14:56:00 (-0700)  main
09.04.18 14:56:00 (-0700)  main
09.04.18 14:56:00 (-0700)  main  X.Org X Server 1.16.4
09.04.18 14:56:00 (-0700)  main  Release Date: 2014-12-20
09.04.18 14:56:00 (-0700)  main  X Protocol Version 11, Revision 0
09.04.18 14:56:00 (-0700)  main  Build Operating System: Linux 4.9.0-4-armmp-lpae armv7l Debian
09.04.18 14:56:00 (-0700)  main  Current Operating System: Linux cb0459c-cb0459c 4.9.59 #2 SMP Wed Apr 4 15:17:24 UTC 2018 armv7l
09.04.18 14:56:00 (-0700)  main  Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1680 bcm2708_fb.fbheight=1050 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
09.04.18 14:56:00 (-0700)  main  Build Date: 16 October 2017  09:26:27AM
09.04.18 14:56:00 (-0700)  main  xorg-server 2:1.16.4-1+deb8u2 (http://www.debian.org/support)
09.04.18 14:56:00 (-0700)  main  Current version of pixman: 0.33.3
09.04.18 14:56:00 (-0700)  main         Before reporting problems, check http://wiki.x.org
09.04.18 14:56:00 (-0700)  main         to make sure that you have the latest version.
09.04.18 14:56:00 (-0700)  main  Markers: (--) probed, (**) from config file, (==) default setting,
09.04.18 14:56:00 (-0700)  main         (++) from command line, (!!) notice,(II) informational,
09.04.18 14:56:00 (-0700)  main         (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
09.04.18 14:56:00 (-0700)  main  (==) Log file: "/var/log/Xorg.0.log", Time:Mon Apr  9 21:56:00 2018
09.04.18 14:56:00 (-0700)  main  (==) Using system config directory "/usr/share/X11/xorg.conf.d"
09.04.18 14:56:11 (-0700)  main  [177:0409/215611.457673:WARNING:audio_manager.cc(293)] Multiple instances of AudioManager detected
09.04.18 14:56:11 (-0700)  main  [177:0409/215611.458306:WARNING:audio_manager.cc(254)] Multiple instances of AudioManager detected
09.04.18 14:56:12 (-0700)  main  [177:0409/215612.729990:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
09.04.18 14:56:12 (-0700)  main  [177:0409/215612.732822:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
09.04.18 14:56:12 (-0700)  main  [177:0409/215612.734807:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
09.04.18 14:56:12 (-0700)  main  [177:0409/215612.747641:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
09.04.18 14:56:12 (-0700)  main  [177:0409/215612.749737:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
```

gibbsonn, 2018-04-23:
I'm getting the same issue as well, same pi version

```
23.04.18 18:25:04 (+0100) X.Org X Server 1.18.4
23.04.18 18:25:04 (+0100) Release Date: 2016-07-19
23.04.18 18:25:04 (+0100) X Protocol Version 11, Revision 0
23.04.18 18:25:04 (+0100) Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
23.04.18 18:25:04 (+0100) Current Operating System: Linux 4056ff1 4.9.50 #2 SMP Wed Nov 1 16:03:39 UTC2017 armv7l
23.04.18 18:25:04 (+0100) Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000 dwc_otg.lpm_enable=0 console=null root=/dev/mmcblk0p2 rootfstype=ext4 rootwait vt.global_cursor_default=0
23.04.18 18:25:04 (+0100) Build Date: 11 November 2016  11:59:59AM
23.04.18 18:25:04 (+0100) xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
23.04.18 18:25:04 (+0100) Current version of pixman: 0.33.3
23.04.18 18:25:04 (+0100)       Before reporting problems, check http://wiki.x.org
23.04.18 18:25:04 (+0100)       to make sure that you have the latest version.
23.04.18 18:25:04 (+0100) Markers: (--) probed, (**) from config file, (==) default setting,
23.04.18 18:25:04 (+0100)       (++) from command line, (!!) notice, (II) informational,
23.04.18 18:25:04 (+0100)       (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
23.04.18 18:25:04 (+0100) (==) Log file: "/var/log/Xorg.0.log", Time: Mon Apr 23 17:25:04 2018
23.04.18 18:25:04 (+0100) (==) Using system config directory "/usr/share/X11/xorg.conf.d"
23.04.18 18:25:10 (+0100) Application is already running 'registry2.resin.io/v2/99aff053459dd9b3fd9f9f872d7d176b'
23.04.18 18:25:23 (+0100) [187:0423/172523.770681:WARNING:audio_manager.cc(293)] Multiple instances ofAudioManager detected
23.04.18 18:25:23 (+0100) [187:0423/172523.771234:WARNING:audio_manager.cc(254)] Multiple instances ofAudioManager detected
23.04.18 18:25:25 (+0100) [187:0423/172525.438505:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
23.04.18 18:25:25 (+0100) [187:0423/172525.438830:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
23.04.18 18:25:25 (+0100) [187:0423/172525.439068:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
23.04.18 18:25:25 (+0100) [187:0423/172525.439239:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
23.04.18 18:25:25 (+0100) [187:0423/172525.439379:ERROR:browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process.
```

Anyone know what's causing this?

Sega01, 2018-04-25:
Following worked for me: 

- changed dependencies to: "electron": "~1.6.16" and "electron-rebuild": "^1.7.2"
- in main.js add app.disableHardwareAcceleration(); to get rid of the browser_gpu_channel_host_factory.cc(103)] Failed to launch GPU process. error

mmranger, 2018-05-02:
also v`2.0.0-beta.8` works (with added bonus of loading pdf's in iframes) see https://github.com/resin-io/resin-electronjs/issues/70