balena-io-experimental/balena-electronjs #70: this boilerplate fails to load when updated to electron latest version 1.8.4

tony1377, 2018-04-24:
steps to reproduce

* do fresh checkout of this repo
* update `electron` to latest (1.8.4) at time of writing
* deploy to a raspberry pi3 model B, running on a pi 7 inch touchscreen

result = blank white screen
expected = the hello world message with unicorn

log output from resin

```
24.04.18 13:56:55 (+0100) Installed application 'registry2.resin.io/v2/e82f07a02fcd0ea86ac979eb102db75a'
24.04.18 13:56:55 (+0100) Starting application 'registry2.resin.io/v2/e82f07a02fcd0ea86ac979eb102db75a'
24.04.18 13:56:56 (+0100) Started application 'registry2.resin.io/v2/e82f07a02fcd0ea86ac979eb102db75a'
24.04.18 13:56:56 (+0100) Systemd init system enabled.
24.04.18 13:56:57 (+0100) systemd 215 running in system mode. (+PAM +AUDIT +SELINUX +IMA +SYSVINIT +LIBCRYPTSETUP +GCRYPT +ACL +XZ -SECCOMP -APPARMOR)
24.04.18 13:56:57 (+0100) Detected virtualization 'other'.
24.04.18 13:56:57 (+0100) Detected architecture 'arm'.
24.04.18 13:56:57 (+0100) Set hostname to <4056ff1>.
24.04.18 13:56:59 (+0100) hostname: No address associated with hostname
24.04.18 13:56:59 (+0100) xauth:  file /root/.Xauthority does not exist
24.04.18 13:56:59 (+0100) xauth: (stdin):1:  bad display name "4056ff1:0" in "add" command
24.04.18 13:56:59 (+0100)
24.04.18 13:56:59 (+0100)
24.04.18 13:56:59 (+0100) X.Org X Server 1.18.4
24.04.18 13:56:59 (+0100) Release Date: 2016-07-19
24.04.18 13:56:59 (+0100) X Protocol Version 11, Revision 0
24.04.18 13:56:59 (+0100) Build Operating System: Linux 4.4.26-v7+ armv7l Raspbian
24.04.18 13:56:59 (+0100) Current Operating System: Linux 4056ff1 4.9.50 #2 SMP Wed Nov 1 16:03:39 UTC2017 armv7l
24.04.18 13:56:59 (+0100) Kernel command line: 8250.nr_uarts=0 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000 dwc_otg.lpm_enable=0 console=null root=/dev/mmcblk0p2 rootfstype=ext4 rootwait vt.global_cursor_default=0
24.04.18 13:56:59 (+0100) Build Date: 11 November 2016  11:59:59AM
24.04.18 13:56:59 (+0100) xorg-server 2:1.18.4-2+rpi1 (https://www.debian.org/support)
24.04.18 13:56:59 (+0100) Current version of pixman: 0.33.3
24.04.18 13:56:59 (+0100)       Before reporting problems, check http://wiki.x.org
24.04.18 13:56:59 (+0100)       to make sure that you have the latest version.
24.04.18 13:56:59 (+0100) Markers: (--) probed, (**) from config file, (==) default setting,
24.04.18 13:56:59 (+0100)       (++) from command line, (!!) notice, (II) informational,
24.04.18 13:56:59 (+0100)       (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
24.04.18 13:56:59 (+0100) (==) Log file: "/var/log/Xorg.0.log", Time: Tue Apr 24 12:56:59 2018
24.04.18 13:56:59 (+0100) (==) Using system config directory "/usr/share/X11/xorg.conf.d"
24.04.18 13:57:05 (+0100) [200:0424/125705.905449:WARNING:audio_manager.cc(293)] Multiple instances ofAudioManager detected
24.04.18 13:57:05 (+0100) [200:0424/125705.905930:WARNING:audio_manager.cc(254)] Multiple instances ofAudioManager detected
```

mmranger, 2018-05-01:
Hey, cannot find the original issue with electron, but there was an issue with the render starting; using electron version 1.8 on RPI 3.

To resolve we went with the `2.0.0-beta.8` version, but we also had to install `libgtk-3.0` instead of `libgtk2.0-dev`.

We just made these changes and thus far have not found any other issues using a beta version of electron.
Hope this helps.

RyanDevlin, 2018-05-04:
@mmranger , I apologize if this is a stupid question, I'm pretty new to all of this. I was wondering how you updated to electron 2.0.0 in this repo. I tried changing the electron dependency in the package.json file from "~1.6.16" to "2.0.0" but this caused the application to crash when launched. Is there something I'm missing? Thanks.

curcuz, 2018-05-09:
this issue is now high prio since the currently pinned electron version is affected by a known vulnerability https://nvd.nist.gov/vuln/detail/CVE-2018-1000118

curcuz, 2018-05-09:
the current approach I am pursuing is to bump electron to `1.7.13` which fixes the vulnerability. From there, we can test later versions too. Please keep in mind that since electron object properties and APIs might change across versions, that's why we pin a specific one