balena-io-experimental/balena-electronjs #90: RPi touch display not working anymore

martenjacobs, 2019-05-24:
I'm using this boilerplate to develop an app that will run on a balenaFin board with an official Raspberry Pi Touch Display connected. This worked fine before, but today I updated my repo to match the boilerplate in the current version of this repo because some apt repos were not available anymore so the old Dockerfile didn't build anymore. Now, the screen works fine, but there aren't any touch events registered anymore. My previous Dockerfile was an exact match with [this one](https://github.com/balena-io/resin-electronjs/blob/7d976ce44d7ecd5a91cda8bf798905b021d7841b/Dockerfile.template), and now I'm using [this one](https://github.com/balena-io/resin-electronjs/blob/5bd034711f7f308fb9a5123fe638fe39e061f6f0/Dockerfile.template). Reverting to my previous commit solves the issue, so it's definitely not hardware related.

martenjacobs, 2019-05-24:
This issue may be the same as [this one](https://github.com/balena-os/balena-raspberrypi/issues/300), but as I can get it working with an older version of the container, it seems the issue is actually related to the container rather than the OS

danielmahon, 2019-05-24:
@martenjacobs try adding `ENV UDEV 1` to your Dockerfile

martenjacobs, 2019-05-25:
@danielmahon That works! Can you explain why this is necessary?

danielmahon, 2019-05-26:
@martenjacobs Honestly... I'm not entirely sure 😄. I'd wager it flags something during the image startup to enable udev so that devices (like a touchscreen) are accessible from the container. But one of the Balena peeps will need to chime in here (@shaunmulligan?) because I think it has to do with their supervisor as I couldn't find much info on the flag outside of Balena forums.

martenjacobs, 2019-05-26:
Yeah I think I read somewhere it enables UDEV (which is kind of in the name, really). I just didn't expect udev to be required for these inputs because they would be handled by the host OS, but I guess that's not the case.

danielmahon, 2019-06-01:
@martenjacobs Just came across this in their docs:
https://www.balena.io/docs/reference/base-images/base-images/#working-with-dynamically-plugged-devices

> ## Working with Dynamically Plugged Devices

> In many IoT projects your containers will want to interact with some hardware, often this hardware is plugged in at runtime, in the case of USB or serial devices. In these cases you will want to enable udevd in your container. In balenalib images this can easily be done either by adding ENV UDEV=1 in your Dockerfile or by setting an environment variable.

> You will also need to run your container privileged, by default any balenaCloud projects that don't contain a docker-compose.yml will run their containers privileged. If you are using a multicontainer project you will need to add privileged: true to each of the service definitions for the services that need hardware access.

martenjacobs, 2020-06-12:
@danielmahon I just made the same mistake again and re-discovered this solution. You solved my problem again :)

closed 2020-06-12 by martenjacobs

danielmahon, 2020-06-12:
@martenjacobs haha, I once found the perfect answer to a problem I was having on Stack Overflow... only to find out I was the one that posted the answer years ago in the first place... I felt proud and stupid at the same time.