balena-io-experimental/balena-electronjs #94: Raspberry Pi 4 Compatibility

cjpollock, 2019-09-23:
Hi team,

When trying to compile this with Balena on a Pi 4, I get the following error:

`E: Unable to locate package libgnome-keyring-dev
[electron]
[electron]      Removing intermediate container 57686fd91418
[electron]      The command '/bin/sh -c install_packages   apt-utils   clang   libxcb-image0   libxcb-util0   xdg-utils   libdbus-1-dev   libgtk2.0-dev   libnotify-dev   libgnome-keyring-dev   libgconf2-dev   libasound2-dev   libcap-dev   libcups2-dev   libxtst-dev   libxss1   libnss3-dev   libsmbclient   libssh-4   libexpat-dev' returned a non-zero code: 10`

Any thoughts?

floion, 2019-09-23:
@nghiant can you check of the package reported missing above should be in the base image?

imrehg, 2019-09-24:
This is not Pi4 incompatibility, but buster incompatibility, even switching to other device types it works like this. Thus need an overhaul for Buster

imrehg, 2019-09-24:
Also can miss `libgnome-keyring0`.

imrehg, 2019-09-24:
It might need `gnome-keyring` instead https://packages.debian.org/search?keywords=gnome-keyring&searchon=names&suite=stable&section=all

lurch, 2019-09-24:
Until this is fixed, would it be worth modifying this repo to use the previous version of Debian? :man_shrugging:

yusbel80, 2020-01-27:
Any update on how to get this working?

lurch, 2020-01-27:
I've not tried it myself, but a comment in #95 says that it works on Pi4.

yusbel80, 2020-01-28:
> I've not tried it myself, but a comment in #95 says that it works on Pi4.

Nah this isn't working, at least its not stable.

I am getting error
[image]

lurch, 2020-01-28:
I believe this issue was originally about a problem at build-time, but it seems like you're having a problem at run-time? So maybe it's a separate problem which deserves a separate issue? :man_shrugging: