balena-io-experimental/balena-electronjs #97: Build fails on RPi4  ( E: Unable to locate package libgnome-keyring-dev )

Craigson, 2019-12-13:
Attempting to build on the RPi4 fails for apt-get packacge `libgnome-keyring-dev`.

Stack trace:
```
[Info]     Building on arm01
[Info]     Pulling previous images for caching purposes...
[Success]  Successfully pulled cache images
[main]     Step 1/12 : FROM balenalib/raspberrypi4-64-node:10-build as build
[main]      ---> c39596a1efef
[main]     Step 2/12 : WORKDIR /usr/src/app
[main]      ---> Running in eb96fc6f5a89
[main]     Removing intermediate container eb96fc6f5a89
[main]      ---> 867e5f3800b1
[main]     Step 3/12 : RUN install_packages   apt-utils   clang   libxcb-image0   libxcb-util0   xdg-utils   libdbus-1-dev   libgtk2.0-dev   libnotify-dev   libgnome-keyring-dev   libgconf2-dev   libasound2-dev   libcap-dev   libcups2-dev   libxtst-dev   libxss1   libnss3-dev   libsmbclient   libssh-4   libexpat-dev
[main]      ---> Running in 764131c89526
[main]     Reading package lists...
[main]     Building dependency tree...
[main]     Reading state information...
[main]     E: Unable to locate package libgnome-keyring-dev
[main]
[main]     apt failed, retrying
[main]     Reading package lists...
[main]     Building dependency tree...
[main]     Reading state information...
[main]     E: Unable to locate package libgnome-keyring-dev
[main]
[main]     apt failed, retrying
[main]     Reading package lists...
[main]     Building dependency tree...
[main]     Reading state information...
[main]     E: Unable to locate package libgnome-keyring-dev
[main]
[main]     Removing intermediate container 764131c89526
[main]     The command '/bin/sh -c install_packages   apt-utils   clang   libxcb-image0   libxcb-util0   xdg-utils   libdbus-1-dev   libgtk2.0-dev   libnotify-dev   libgnome-keyring-dev   libgconf2-dev   libasound2-dev   libcap-dev   libcups2-dev   libxtst-dev   libxss1   libnss3-dev   libsmbclient   libssh-4   libexpat-dev' returned a non-zero code: 100
[Info]     Uploading images
[Success]  Successfully uploaded images
[Error]    Some services failed to build:
[Error]      Service: main
[Error]        Error: The command '/bin/sh -c install_packages   apt-utils   clang   libxcb-image0   libxcb-util0   xdg-utils   libdbus-1-dev   libgtk2.0-dev   libnotify-dev   libgnome-keyring-dev   libgconf2-dev   libasound2-dev   libcap-dev   libcups2-dev   libxtst-dev   libxss1   libnss3-dev   libsmbclient   libssh-4   libexpat-dev' returned a non-zero code: 100
[Info]     Built on arm01
[Error]    Not deploying release.
Remote build failed

Additional information may be available by setting a DEBUG=1 environment
variable: "set DEBUG=1" on a Windows command prompt, or "export DEBUG=1"
on Linux or macOS.

If you need help, don't hesitate in contacting our support forums at
https://forums.balena.io

For bug reports or feature requests, have a look at the GitHub issues or
create a new one at: https://github.com/balena-io/balena-cli/issues/
```

lurch, 2019-12-16:
Duplicate of #94 ?

kfadgen, 2020-05-30:
Is there any update on this?  I'm seeing the same issue here.

lurch, 2020-05-31:
https://github.com/balena-io/balena-electronjs/pull/95#issuecomment-536414333